esp-rs/esp-println #14: Use USB-CDC on esp32s2

UseAfterFreee, 2022-10-22:
I might be misunderstanding something (or doing something wrong) as I am quite new to the esp-rs community, so sorry for that in advance. I currently have some [WEMOS lolin 32 s2 mini](https://www.wemos.cc/en/latest/s2/s2_mini.html) laying around and wanted to try creating a no-std rust project for them. For simple logging purposes I wanted to use the [USB Serial Console (CDC)](https://docs.espressif.com/projects/esp-idf/en/v4.3/esp32s2/api-guides/usb-console.html), as I do not currently have a jtag or uart to usb adapter laying around. Is it possible to use the CDC with this crate? And if not, is this planned to be implemented?

jessebraham, 2022-10-22:
It is not currently possible to use USB CDC with this crate, but that's something that we should add support for. To be honest I just learned of  this feature recently when it was added to `espflash`, so it was just missed here :)

I should be able to take a look into this next week some time.

joeatbayes, 2024-03-15:
Was this ever implemented for the EPS32-s2?  if so can you please provide a link for how to configure it?

MabezDev, 2024-03-17:
It's not implemented yet.

jath03, 2024-04-28:
I'm coming from https://github.com/esp-rs/espflash/issues/621, and am researching how to implement this.  According to [the espressif docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/usb-otg-console.html), there is a USB CDC implementation in the ROM code.  I'm trying to look at the [ROM sources](https://github.com/espressif/esp-idf/blob/master/components/esp_rom/include/esp32s2/rom/usb/usb_cdc.h) to understand how to use it, but I'm not getting very far.  I could use some pointers from someone with a bit more experience with this stuff.  I don't really know what I'm looking at with the ROM stuff.

bjoernQ, 2024-04-29:
You could probably search for `CONFIG_ESP_CONSOLE_USB_CDC` in ESP-IDF. Especially https://github.com/espressif/esp-idf/blob/master/components/bootloader_support/src/bootloader_console.c might be interesting