esp-rs/esp-println #20: Use `esp-println` as a defmt logger

TheButlah, 2022-11-20:
I wanted to build on `esp-println` to implement a `defmt` global logger, since `defmt` is great and esp support for `defmt` is still nascent.

I ended up making [`defmt_esp_println`](https://github.com/SlimeVR/SlimeVR-Rust/tree/86f6100b859ff6cef2de7d259999b06335908294/firmware/crates/defmt_esp_println), and thought it might be helpful for others too, so I wanted to create this issue. Was unsure if this would make more sense to live inside this repo or the `esp-rs` organization or not. Feel free to close it if its not helpful or too unrelated.

bjoernQ, 2022-11-21:
NIce to see you managed to implement this! I guess it might be definitely useful for others. I also thought about implementing something like this but it was at the very bottom of my TODO list

I'm not sure if this should be part of esp-println itself - I think it's totally fine to have this as a separate crate.
Also not sure if it should be a `esp-rs` crate - in any case you can add it to https://github.com/esp-rs/awesome-esp-rust

bugadani, 2023-08-26:
I probably should have looked around first...

PR #48 (2023-08-29): Add defmt backend
 - Add `defmt` feature to enable the backend
 - Backend injects 2+2 framing bytes so we don't require rzcobs for framing

Closes #20
closed 2023-08-29 by bjoernQ