esp-rs/esp-println #26: Questions about soundness of code

TheButlah, 2022-11-22:
Given that `Printer` can essentially be created as many times as we want by virtue of it not being a singleton, `&mut self` doesn't actually guarantee against concurrent access to the registers being operated on. So why is this not unsound? 
I could see it being sound if the critical section was always present, but [its not, depending on the feature flags](https://github.com/esp-rs/esp-println/blob/00baa103e32af1a66d88efa1896fa4565c82868e/src/lib.rs#L170?). 

I know that in single core systems the volatile writes/reads *probably* are atomic so that should free of data races, but what about multicore systems like the esp32 and esp32s3? And if we take a very pedantic view on it, [volatile writes are != atomic writes](https://doc.rust-lang.org/std/ptr/fn.write_volatile.html#safety) so this is still undefined behavior and could be a data race.

Furthermore I would expect that if I am writing to a peripheral, there would be some state in it that has some state sensitive to race conditions, such as the FIFO queues etc, racy not just in terms of "my data is being interleaved" but rather actually causing the peripheral to be in an invalid state (I'm less sure about this because I'm not sure if any of these peripherals would actually have race conditions like that.

My guess for the first part of my question is that even though rust considers concurrent

bjoernQ, 2022-11-22:
In the very beginning this was really only using ROM functions which took care of concurrency issues - it was possible to print gibberish but nothing fatal could happen. Back then the idea was that the user needs to take care to get useable output

Now, especially on S2 where we don't use the ROM function (and in general for SerialJtag etc) this can lead to issues including a complete stall of the system - one of the reasons the critical-section feature was added. I think the reason to not make it default was that it's a behavior change

I'm fine with promoting the critical-section feature to a default feature

TheButlah, 2022-11-22:
Ultimately then, what are your thoughts on making the critical section approach the *only* option?

bjoernQ, 2022-11-22:
I think it's still good give the user a choice -maybe it's worth a paragraph in the README but if someone really wants to opt-out it should be possible

TheButlah, 2022-11-22:
I'm that case, I wonder if the feature flag should go the other way around - we activate a feature to *disable* the critical section, and call it unsafe-something.

The reason why we would do that is because libraries will depend on the default features by default, and once you have a feature activated by a dependency, you can't deactivate it.

So applications would turn on the feature, but *not* libraries.

TheButlah, 2022-11-22:
Alternatively (or more likely in addition to?) You can have unsafe versions of the functions that don't have critical sections. I could call those in my deft-esp-println since defmt already will use critical sections so there is no point to nesting more of them.

That would mean that even if the application decided to keep the library as it's default safe mode, I would have opted into the enhanced speed because I'm using the unsafe functions in a sound way in my library

bjoernQ, 2022-11-22:
Not sure about having an opt-out feature. I think using this as a dependency of a library is also an edge-case.

Usually, libraries would use the `log` crate and the application decides what happens to the log messages. They can use this crate with the log feature or whatever fits them

But offering an unsafe interface which never uses critical-sections for e.g. your defmt crate certainly makes sense

MabezDev, 2024-05-23:
The critical section feature has been added and is enabled by default, so I think we can close this now.

closed 2024-05-23 by MabezDev