esp-rs/esp-println #44: ESP8266 cannot format args

knightpp, 2023-06-24:
Hi. It seems `esp-println` cannot fully support ESP8266 or something wrong with my setup or board. I know that ESP8266 is quite old and is not actively supported but decided to create issue if someone else will encounter this issue.

## println!
Trying to print with `println!("any string")` works fine, but when I use formatting string to actually add some variables, the board does not print any output e.g. `println!("delay {}ms", 1u8)`.

If I change the print statement to `println!("delay {}ms", -1i8);` I see the following output repeatedly:
```
Fatal exception (2): 
epc1=0x3ffe848c, epc2=0x00000000, epc3=0x00000000, excvaddr=0x3ffe848c, depc=0x00000000
0x3ffe848c - _data_end
    at ??:??
0x00000000 - XT_STK_PC
    at ??:??
0x00000000 - XT_STK_PC
    at ??:??
0x3ffe848c - _data_end
    at ??:??
0x00000000 - XT_STK_PC
    at ??:??
```
But if I flash the board with `cargo run --release` it works in both cases above.

## dbg!
And final observation, using `dbg!` causes errors. 
```rust
    let mut i = 0u8;
    loop {
        i += 1;
        dbg!(i);
        timer1.delay_ms(100);
    }
```
If using `cargo run` will output rubbish indefinitely.
And using `cargo run --release` will output the following error repeatedly
```
Fatal exception (2): 
epc1=0x3ffe8050, epc2=0x00000000, epc3=0x00000000, excvaddr=0x3ffe8050, depc=0x00000000
0x3ffe8050 - _rodata_start
    at ??:??
0x00000000 - XT_STK_PC
    at ??:??
0x00000000 - XT_STK_PC
    at ??:??
0x3ffe8050 - _rodata_start
    at ??:??
0x00000000 - XT_STK_PC
    at ??:??
```

## Cargo.toml and main.rs
`main.rs`
```rust
#![no_std]
#![no_main]

use esp8266_hal::prelude::*;
use esp8266_hal::target::Peripherals;
use esp_println::{dbg, print, println};
use panic_halt as _;

#[entry]
fn main() -> ! {
    let dp = Peripherals::take().unwrap();
    let (mut timer1, _) = dp.TIMER.timers();

    loop {
        println!("main loop delaying for {}ms", 1u8);
        // println!("a");
        // dbg!("hello");
        timer1.delay_ms(1500);
    }
}
```

`Cargo.toml`
```rust
[package]
name = "esp8266"
version = "0.1.0"
edition = "2021"
readme = "README.md"

[dependencies]
esp-println = { version = "0.5.0", default-features = false, features = [
    "esp8266",
    "uart",
] }
esp8266-hal = "0.5.1"
panic-halt = "0.2.0"

[profile.dev]
lto = true
opt-level = "z"
debug = false
overflow-checks = false

[profile.release]
lto = true
opt-level = "s"
```

jessebraham, 2023-06-27:
Thanks for the issue.

I have played around a bit with `esp8266-hal`, and other than needing to update some dependencies and add a `CriticalSection` implementation it I was able to get it working with `esp_println` without issue. I tested all sorts of formatting, wasn't able to trigger an exception.

I can possibly try to update `esp8266-hal` just to bring it back up to date, but this project is (unfortunately) incredibly low-priority for us right now, so I can't make any promises.

t-moe, 2023-09-13:
In my case println! with args works fine on the esp8266, without crashing.

e.g. even  `println!("Hello world 1! {} {}", 13, format_args!("a {} b {}", 1, 2));` works as expected and outputs `Hello world 1! 13 a 1 b 2`
However, using the log macros from the log crate (e.g. `warn!`), the argument formatting does not work.
e.g. ` warn!("Logger is setup {}", 28);` just prints `WARN - `. (It looks like the `record.args()` are emtpy?)

jessebraham, 2024-02-05:
We have decided to drop support for the ESP8266 (see #82), so closing this issue.

closed 2024-02-05 by jessebraham