esp-rs/esp-println #51: ESP32 stuck in ROM function

raphaelhetzel, 2023-10-10:
When using the library (0.6) from an embassy task (using log::info!), my program always gets stuck on https://github.com/esp-rs/esp-println/blob/main/src/lib.rs#L173 . I've tried this with the critical-section flag enabled and disabled. https://github.com/espressif/esp-idf/blob/master/components/esp_rom/include/esp32/rom/uart.h#L230 suggests that the function will block until the FIFO (guess this is defined in the ROM) is empty. Are you aware of how that FIFO is serviced?  I'm trying to pin down where I might be causing this to block. My suspects are currently either embassy-futures, embassy-sync (channels) or esp-wifi.

I also saw that there is a workaround for the S2, what type of flaky did you experience that required that fix? I haven't looked into the details of that fix but that seems like you're bypassing the UART driver but I did not spot where you got the 0x3f400004 and 0x3f400010 addresses from.

Thanks for any hints you can provide on this.

bjoernQ, 2023-10-10:
We recently had a fix (https://github.com/esp-rs/esp-hal/pull/823)  for the linker script on ESP32 which might be worth to try. I guess you are using the currently release HAL?

The fix isn't yet in the released version yet

raphaelhetzel, 2023-10-10:
Yes, going from esp-hal 0.15 to the current main branch fixes the issue.
Thanks for that!

Interestingly both have a different (broken) output in the beginning of the main function: This isn't an issue for me but might hint at some rom component still not being used as intended.

Code:
```
#[entry]
fn main() -> ! {
    esp_println::logger::init_logger_from_env();
    esp_println::println!("Start Edgeless Embedded.");
```

0.15:
`St?*������́Embedded.`

Current Main:
`S�сEdgeless Embedded.`

bjoernQ, 2023-10-11:
Doesn't seem to happen to me. Do you have a full example to reproduce this?

raphaelhetzel, 2023-10-11:
The minimal example seems to be a combination of using the clock control and allocating a huge chunk of memory after the log statement:

```
#[entry]
fn main() -> ! {
    esp_println::logger::init_logger_from_env();
    esp_println::println!("Start Edgeless Embedded.");
 
    const HEAP_SIZE: usize = 32 * 1024;
    static mut HEAP: core::mem::MaybeUninit<[u8; HEAP_SIZE]> = core::mem::MaybeUninit::uninit();
    
    let peripherals = hal::peripherals::Peripherals::take();
    let mut system = peripherals.SYSTEM.split();
    
    let clocks =  hal::clock::ClockControl::max(system.clock_control).freeze();
}
```

with the current main, this leads to
`S�.*������́Embedded.`

setting up esp-alloc with the heap leads to the result described above.

bugadani, 2023-11-07:
The hardware may still be printing when you switch the clock under it. I don't think this is an issue with esp-println, or one that can be solved here except by taking a frozen Clocks struct as initialization which would just be painful IMO.

bjoernQ, 2024-01-15:
I guess we can close this. Feel free to re-open with more information if not

closed 2024-01-15 by bjoernQ