esp-rs/esp-println #52: defmt not working as expected

SergioGasquez, 2023-10-19:
I've generated a template project (`cargo generate esp-rs/esp-template --name test -d mcu=esp32c3 -d advanced=false`), updated the `esp-println` feature to `esp-println = { git = "https://github.com/esp-rs/esp-println", features = ["esp32c3", "defmt"]  }`, included the `defmt` crate and tried several things:
- If we use `espflash v2.1.0` which does not have `defmt` support:
  - `esp_println::println!("Hello world!")` prints the message
  - Using any `defmt` macro results in a linker error, which I guess is expected
- If we use `espflash v2.1.1-dev` (`main` branch) with `defmt` feature
  - `defmt::println!("Hello world!")` results in a linker error
  - `defmt::debug!("Hello world!")` builds but does not print the message, same for `defmt::info` and `defmt::warn`
  - `defmt::error!` results in a linker error
   - `esp_println::println` prints the message fine
- If we use `espflash v2.1.1-dev` (`main` branch) with `defmt` feature
  - `defmt::println!("Hello world!")` results in a linker error
  - `defmt::debug!("Hello world!")` builds but does not print the message, same for `defmt::info` and `defmt::warn`
  - `defmt::error!` results in a linker error
  - `esp_println::println` prints the message fine
Using the `defmt-raw` instead of the `defmt` feature in `esp-println` results in similar outputs.

cc @bugadani

SergioGasquez, 2023-10-19:
> you need to set DEFMT_LOG to some level to print messages above that level

Did some quick testing, trying to print a defmt::info! with `DEFMT_LOG=warn` does not print the message, with `DEFMT_LOG=info`results in linker script error

bugadani, 2023-10-19:
[image]

works as expected. you'll need to add to your `.cargo/config.toml` this line:

```
[target.xtensa-esp32s3-none-elf]
rustflags = [
  "-C", "link-arg=-nostartfiles",
  "-C", "link-arg=-Wl,-Tlinkall.x",
  "-C", "link-arg=-Trom_functions.x",
  "-C", "link-arg=-Tdefmt.x", <--- this
]
```

SergioGasquez, 2023-10-19:
Still not working for me, I also tried with S3:
1. `cargo generate esp-rs/esp-template --name=defmt -d mcu=esp32s3 -d advanced=false`
2. Modified .cargo/config.toml:
```tomlç[target.xtensa-esp32s3-none-elf]
runner = "espflash flash --monitor"

[build]
rustflags = [
  "-C",
  "link-arg=-Tlinkall.x",
  "-C",
  "link-arg=-nostartfiles",
  "-C",
  "link-arg=-Tdefmt.x",
  "-C",
  "link-arg=-Trom_functions.x",
]

target = "xtensa-esp32s3-none-elf"

[unstable]
build-std = ["core"]
```
3. Modified Cargo.toml, added: 
```
esp-println = { git = "https://github.com/esp-rs/esp-println", features = [
    "esp32s3",
    "defmt-raw",
] }
defmt = "0.3.5"
```
4. Removed `esp_println::prinln!` calls in `main.rs` and added `defmt::info!("Hello world!");`
5. `DEFMT_LOG=info cargo r` results in linker error

bugadani, 2023-10-19:
Please don't use `defmt-raw`. It's meant as an advanced feature for folks who _only_ use defmt and have the tools to read it. espflash users are supposed to use `defmt` instead.

> cannot open linker script file rom_functions.x

I can reproduce this error, but this is an esp-wifi file and is not necessary for defmt. Remove the 
```
  "-C",
  "link-arg=-Trom_functions.x",
```

part and the project compiles.

SergioGasquez, 2023-10-19:
> Please don't use defmt-raw. It's meant as an advanced feature for folks who only use defmt and have the tools to read it. espflash users are supposed to use defmt instead.

Ups, that was a typo, let me try without rom functions

SergioGasquez, 2023-10-19:
Still getting linker error or nothing printed. 
- main.rs:
```rust
#![no_std]
#![no_main]

use esp_backtrace as _;
use hal::{clock::ClockControl, peripherals::Peripherals, prelude::*, Delay};

#[entry]
fn main() -> ! {
    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();
    let clocks = ClockControl::max(system.clock_control).freeze();
    let mut delay = Delay::new(&clocks);

    // println!("Hello world!");
    defmt::warn!("Hello world!");
    loop {
        defmt::info!("Hello world!");

        // println!("Loop...");
        delay.delay_ms(500u32);
    }
}
```
- Cargo.toml
```toml
[package]
name    = "defmt"
version = "0.1.0"
authors = ["Sergio Gasquez <sergio.gasquez@gmail.com>"]
edition = "2021"
license = "MIT OR Apache-2.0"

[dependencies]
hal = { package = "esp32s3-hal", version = "0.12.0" }
esp-backtrace = { version = "0.8.0", features = [
    "esp32s3",
    "panic-handler",
    "exception-handler",
    "print-uart",
] }
esp-println = { git = "https://github.com/esp-rs/esp-println", features = [
    "esp32s3",
    "defmt",
] }
defmt = "0.3.5"
```
- .cargo/config.toml
```toml
[target.xtensa-esp32s3-none-elf]
runner = "espflash flash --monitor"

[build]
rustflags = [
  "-C",
  "link-arg=-Tlinkall.x",
  "-C",
  "link-arg=-nostartfiles",
  "-C",
  "link-arg=-Tdefmt.x",
]

target = "xtensa-esp32s3-none-elf"

[unstable]
build-std = ["core"]
```

bugadani, 2023-10-19:
what's the exact linker error you see?

SergioGasquez, 2023-10-19:
```
error: linking with `xtensa-esp32s3-elf-gcc` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/Users/sergio/.rustup/toolchains/esp/lib/rustlib/aarch64-apple-darwin/bin:/opt/homebrew/bin:/opt/homebrew/sbin:/Users/sergio/.rustup/toolchains/esp/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:/Users/sergio/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/Users/sergio/.cargo/bin:/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin:/opt/homebrew/bin:/opt/homebrew/sbin:/Users/sergio/.rustup/toolchains/esp/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:/Users/sergio/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin" VSLANG="1033" "xtensa-esp32s3-elf-gcc" "/var/folders/_w/919x4xzs7zl5kyydzg5nr_v40000gn/T/rustcJVVKyS/symbols.o" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps/defmt-3a7fed31810c4a3b.1t5wwyxzq3xr2t7v.rcgu.o" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps/defmt-3a7fed31810c4a3b.1xatxwufv56mifoz.rcgu.o" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps/defmt-3a7fed31810c4a3b.2c6fd93hfrl0rvnw.rcgu.o" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps/defmt-3a7fed31810c4a3b.2gy6r62l9r3kmg43.rcgu.o" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps/defmt-3a7fed31810c4a3b.2r394gr85xrdftqm.rcgu.o" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps/defmt-3a7fed31810c4a3b.2sxu2r1jifg0jh8.rcgu.o" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps/defmt-3a7fed31810c4a3b.49d8b7bo4b1sxcjn.rcgu.o" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps/defmt-3a7fed31810c4a3b.558ozvhyz9fe7vvb.rcgu.o" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps/defmt-3a7fed31810c4a3b.5d4lnacj13j6361r.rcgu.o" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps/defmt-3a7fed31810c4a3b.k3qmbeojym8xtv1.rcgu.o" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps/defmt-3a7fed31810c4a3b.ksnplor95kgg45l.rcgu.o" "-Wl,--as-needed" "-L" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps" "-L" "/Users/sergio/Documents/Espressif/tests/defmt/target/debug/deps" "-L" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/build/defmt-aaf09b57d34cf050/out" "-L" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/build/esp32s3-hal-1a4075be64451cb7/out" "-L" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/build/esp32s3-hal-1a4075be64451cb7/out" "-L" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/build/esp-hal-common-e2187ba43ff17732/out" "-L" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/build/esp32s3-20acba107c24f96e/out" "-L" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/build/xtensa-lx-ce810a44aefe3fce/out" "-L" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/build/xtensa-lx-rt-97a62a37a804626f/out" "-L" "/Users/sergio/.rustup/toolchains/esp/lib/rustlib/xtensa-esp32s3-none-elf/lib" "-Wl,-Bstatic" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps/libdefmt-caad4a23e598647e.rlib" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps/libbitflags-e85f62d11ff9d0e5.rlib" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps/libesp32s3_hal-9469f43f1260a3ff.rlib" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps/libesp_hal_common-607c2ea011b92472.rlib" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps/libembedded_io-b0d1d948907075a4.rlib" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps/libbitfield-32f9af196384c0ab.rlib" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps/libbitflags-b45032d3cf7f110b.rlib" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps/libcfg_if-26968e6f9c77e082.rlib" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps/libesp32s3-006391992160d3d7.rlib" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps/libstrum-28b65754e511ef0b.rlib" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps/libesp_synopsys_usb_otg-1da1b984205ea9a0.rlib" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps/libvcell-8809acefa7d2ff09.rlib" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps/libusb_device-9de331603759a9cb.rlib" "/Users/sergio/Documents/Espressif/tests/defmt/target/xtensa-esp32s3-none-elf/debug/deps/libcritical_section-88f68fb4e61c8eca.rlib" "/Us [... cut ...]
          collect2: error: ld returned 1 exit status
          
  = note: some `extern` functions couldn't be found; some native libraries may need to be installed or have their path specified
  = note: use the `-l` flag to specify native libraries to link
  = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://doc.rust-lang.org/cargo/reference/build-scripts.html#cargorustc-link-libkindname)

error: could not compile `defmt` (bin "defmt") due to previous error
```

SergioGasquez, 2023-10-19:
Added that line, but the output is still the same

bugadani, 2023-10-19:
I can't reproduce any more build errors but I probably should have run the code, not just build it.

```
❯ with-env [DEFMT_LOG "info"] { cargo +esp r }
   Compiling defmt-macros v0.3.6
   Compiling defmt v0.3.5
   Compiling esp-println v0.6.0 (https://github.com/esp-rs/esp-println#8b52dca7)
   Compiling defmt v0.1.0 (C:\_Hobby\defmt)
    Finished dev [unoptimized + debuginfo] target(s) in 2.66s
     Running `espflash flash --monitor target\xtensa-esp32s3-none-elf\debug\defmt`
[2023-10-19T15:10:17Z INFO ] Serial port: 'COM4'
[2023-10-19T15:10:17Z INFO ] Connecting...
[2023-10-19T15:10:17Z INFO ] Using flash stub
Chip type:         esp32s3 (revision v0.1)
Crystal frequency: 40MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       68:b6:b3:2d:94:7c
App/part. size:    198,848/8,323,072 bytes, 2.39%
[00:00:00] [========================================]      14/14      0x0                                                                                                                                                                                                [00:00:00] [========================================]       1/1       0x8000                                                                                                                                                                                             [00:00:01] [========================================]      77/77      0x10000                                                                                                                                                                                            [2023-10-19T15:10:20Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378c15
0x40378c15 - esp_hal_common::interrupt::xtensa::vectored::handle_interrupts
    at C:\Users\bugad\.cargo\registry\src\index.crates.io-6f17d22bba15001f\esp-hal-common-0.12.0\src\interrupt\xtensa.rs:417
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
0x3fce3818 - _stack_end_cpu0
    at ??:??
load:0x403c9700,len:0x4
0x403c9700 - __user_exception
    at ??:??
load:0x403c9704,len:0xc00
0x403c9704 - __user_exception
    at ??:??
load:0x403cc700,len:0x2eb0
0x403cc700 - __user_exception
    at ??:??
SHA-256 comparison failed:
Calculated: b8f485cf73c8ad59b009a332167b4a854ec774585c7fa428926acc30cc0c964b
Expected: 3499e4149a363c2b1ee4ee9709ca5031f50ee6b595a532f697c8659516763ae3
Attempting to boot anyway...
entry 0x403c9908
0x403c9908 - __user_exception
    at ??:??
I (46) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (47) boot: compile time Jun  7 2023 08:07:32
I (47) boot: Multicore bootloader
I (51) boot: chip revision: v0.1
I (55) boot.esp32s3: Boot SPI Speed : 40MHz
I (60) boot.esp32s3: SPI Mode       : DIO
I (65) boot.esp32s3: SPI Flash Size : 8MB
I (69) boot: Enabling RNG early entropy source...
I (75) boot: Partition Table:
I (78) boot: ## Label            Usage          Type ST Offset   Length
I (86) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (93) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (101) boot:  2 factory          factory app      00 00 00010000 007f0000
I (108) boot: End of partition table
I (113) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=06d94h ( 28052) map
I (128) esp_image: segment 1: paddr=00016dbc vaddr=00000001 size=00008h (     8)
I (129) esp_image: segment 2: paddr=00016dcc vaddr=40378000 size=01888h (  6280) load
I (140) esp_image: segment 3: paddr=0001865c vaddr=00000000 size=079bch ( 31164)
I (154) esp_image: segment 4: paddr=00020020 vaddr=42000020 size=20874h (133236) map
I (188) boot: Loaded app from partition at offset 0x10000
I (189) boot: Disabling RNG early entropy source...
[WARN] - Hello world!
[INFO] - Hello world!
[INFO] - Hello world!
[INFO] - Hello world!
[INFO] - Hello world!
[INFO] - Hello world!
```

Output stops after that, and I'm not 100% sure but I suspect this is an espflash issue. I have experienced similar when printing the same string over and over and over again. This doesn't really happen in my nontrivial project, though.

bugadani, 2023-10-19:
Actually, I think you found a bug in `Delay`. Removing that means this demo prints Hello world! continuously, not stopping after a few.

bugadani, 2023-10-19:
Could you please try this? https://github.com/bugadani/esp32s3-defmt-example

I have been able to use this exact project with my USB-connected S3. This works for me without issues: no stopping, everything printed as I'd expect.

I had to change the printer to use jtag_serial, though, which is actually the setting I moved away _from_ in my project, which is something I don't understand.

SergioGasquez, 2023-10-20:
Just tried your `esp32s3-defmt-example` project and it's also working fine for me!

bjoernQ, 2023-10-20:
> If I use the uart feature, I have no dropped bytes, but I have longer hangs sometimes, and this example project stops working reliably. I have no idea what's going on with that. Maybe @bjoernQ could help debug the byte dropping part, but I don't have a reliable reproduction for that.

Once you have a way to reliably reproduce it, we should create an issue for that. (Maybe also without it)

bugadani, 2023-10-20:
> Once you have a way to reliably reproduce it, we should create an issue for that. (Maybe also without it)

cc #53 - the issue is a bit vague but at least it's recorded now :)

bugadani, 2023-10-23:
I this this issue can be closed. The problems here are not specific to defmt, they relate to how rom functions flush (or not) the tx fifo.

The issue just never surfaced because most log prints end in a newline, which triggers a flush in `uart`, or, in `jtag_serial` the flush is done by hand in esp-println.

closed 2023-10-24 by SergioGasquez