esp-rs/esp-println #59: _rtc_fast_data_end when using jtag-serial to print u64

t-moe, 2023-10-26:
I have a very simple app:
```
#![no_std]
#![no_main]

use esp_backtrace as _;
use esp_println::println;
use hal::entry;
#[entry]
fn main() -> ! {
    esp_println::logger::init_logger_from_env();
    println!("{:#04x}", 0xFFFF_FFFF_FFFF_FFFFu64);
    loop {}
}
```

and a very simple Cargo.toml
```
[package]
#.....

[dependencies]
hal = { package = "esp32c6-hal", version = "0.5.0" }
esp-backtrace = { version = "0.8.0", features = ["esp32c6", "panic-handler", "exception-handler", "print-jtag-serial"] }
esp-println = { version = "0.6.0", features = ["esp32c6", "log", "jtag_serial"], default-features = false}
```

When i type `cargo run` i get the following output:

```  Compiling esp-println v0.6.0
   Compiling esp-backtrace v0.8.0
   Compiling u64print v0.1.0 (/tmp/u64print)
    Finished dev [unoptimized + debuginfo] target(s) in 0.25s
     Running `espflash flash --monitor target/riscv32imac-unknown-none-elf/debug/u64print`
[2023-10-26T13:07:10Z INFO ] Serial port: '/dev/ttyACM0'
[2023-10-26T13:07:10Z INFO ] Connecting...
[2023-10-26T13:07:11Z INFO ] Using flash stub
Chip type:         esp32c6 (revision v0.0)
Crystal frequency: 40MHz
Flash size:        8MB
Features:          WiFi 6, BT 5
MAC address:       40:4c:ca:45:60:2c
App/part. size:    188,880/8,323,072 bytes, 2.27%
[00:00:00] [========================================]      13/13      0x0                                                                                                                
[00:00:00] [========================================]       1/1       0x8000                                                                                                             
[00:00:01] [========================================]      74/74      0x10000                                                                                                            [2023-10-26T13:07:13Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x15 (USB_UART_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x408007fe
0x408007fe - software_reset_cpu
    at ??:??
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
0x4086c410 - __global_pointer$
    at ??:??
load:0x4086e610,len:0x2d68
0x4086e610 - __global_pointer$
    at ??:??
load:0x40875720,len:0x1800
0x40875720 - __global_pointer$
    at ??:??
SHA-256 comparison failed:
Calculated: 377c26481ad6ab0cce508450f353a55f4bd7eb9159aaf85fde3111071f164514
Expected: 0af544a033ab3492852b8232c904c578d5f07c9d4a423a64473f060db374ab32
Attempting to boot anyway...
entry 0x4086c410
0x4086c410 - __global_pointer$
    at ??:??
I (41) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (42) boot: compile time Jun  7 2023 08:02:08
I (43) boot: chip revision: v0.0
I (46) boot.esp32c6: SPI Speed      : 40MHz
I (51) boot.esp32c6: SPI Mode       : DIO
I (56) boot.esp32c6: SPI Flash Size : 8MB
I (61) boot: Enabling RNG early entropy source...
I (66) boot: Partition Table:
I (70) boot: ## Label            Usage          Type ST Offset   Length
I (77) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (84) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  2 factory          factory app      00 00 00010000 007f0000
I (99) boot: End of partition table
I (103) esp_image: segment 0: paddr=00010020 vaddr=42000020 size=25fech (155628) map
I (144) esp_image: segment 1: paddr=00036014 vaddr=40808000 size=00014h (    20) load
I (145) esp_image: segment 2: paddr=00036030 vaddr=42026030 size=07100h ( 28928) map
I (155) esp_image: segment 3: paddr=0003d138 vaddr=40808014 size=0106ch (  4204) load
I (159) boot: Loaded app from partition at offset 0x10000
I (164) boot: Disabling RNG early entropy source...
0xffffffffffffffff
0xffffffff - _rtc_fast_data_end
    at ??:??

```

What is that last line?
[image]

bjoernQ, 2023-10-26:
That is espflash's monitor command trying to resolve the hex value as a symbol in the ELF file. It doesn't find an exact match and prints "something".

You also see that for other lines like
```
load:0x4086e610,len:0x2d68
0x4086e610 - __global_pointer$
    at ??:??
```

You won't see that when using some other serial monitor.

Closing this since it's not an esp-println issue

closed 2023-10-26 by bjoernQ

t-moe, 2023-10-26:
oh, I see.  :grinning: 
Thank you