esp-rs/esp-println #65: `defmt::*` over rtt doesn't work with esp-println

MabezDev, 2023-11-07:
Consider the following loop and output:

```
loop { 
      defmt::println!("Loop... {}", x);
      x += 1;
      delay.delay_ms(500u32);
      if x > 10 {
          defmt::println!("About to use println!");
          esp_println::println!("Loop {}", x);
      }
  }
```

```
Loop... 0
Loop... 1
Loop... 2
Loop... 3
Loop... 4
Loop... 5
Loop... 6
Loop... 7
Loop... 8
Loop... 9
Loop... 10
 WARN session_drop: probe_rs::session: Could not clear all hardware breakpoints: Riscv(AbstractCommand(HaltResume))
 WARN session_drop: probe_rs::session: Failed to deconfigure device during shutdown: Riscv(AbstractCommand(HaltResume))
 INFO probe_rs: Wrote log to "/home/mabez/.local/share/probe-rs/1699374676691.log"
Error: Unrecoverable error while decoding Defmt data and some data may have been lost: Malformed
```

By extension of this issue, esp-backtrace is broken in this configuration because it uses `esp_println::println!`.

bugadani, 2023-11-10:
So, the problem is:

 - esp-backtrace depends on esp-println and uses the `println!` macro extensively
 - esp-println can't simply re-export the defmt macros - esp-backtrace would need to conditionally import defmt
 - non-defmt println! outputs serial data without defmt's formatting, which is a no-no for `defmt-raw`

Possibilities I see to resolve the issue:

 - make esp-backtrace defmt-aware (i.e. add a feature and optional dependency). This isn't ideal for the users
 - merge esp-backtrace and esp-println - this isn't ideal because the crates have different responsibilities
 - move the panic/exception handling code to esp-println, while keeping strictly backtrace-related functionality in esp-backtrace. This is a weird worst of both worlds option.
 - add `print-panic` and `print-exception` features to esp-println, and make esp-println depend on esp-backtrace instead of the other way around. This way, users don't have to depend on esp-backtrace at all and we can move code between them however we like. This allows making esp-backtrace defmt-aware without user-facing feature hell, and also simplifies things as users don't have to take care to configure print drivers on two crates.
 - move esp-backtrace into esp-hal - this would create an esp-hal/esp-println dependency which currently doesn't exist

bjoernQ, 2023-11-10:
I wonder if we need the `defmt-raw` feature at all 🤔 

I guess one would only use it with RTT and probe-rs?
But then one could just use defmt-rtt and not use esp-println at all

Does probe-rs include that host-side backtrace support which probe-run had? Then we wouldn't need esp-backtrace, either. (But I think it was only for cortex-m back then)

If using defmt-raw via one of the serial comms that would mean we'd need a silent bootloader and ROM code boot messages turned off - otherwise it will fail during boot?

bjoernQ, 2023-11-27:
Maybe we could even question RTT support in esp-println? It was very useful in the beginning (because all the other crates had some problems on ESP32-C3 back then and IIRC they were unmaintained at that time) but it seems things changed for the better now (or am I wrong?)

MabezDev, 2023-11-27:
I agree, I think my only hesitation is that we won't get panic messages when defmt-rtt; we need panic-probe to support RISCV/Xtensa.

bjoernQ, 2023-11-27:
> I agree, I think my only hesitation is that we won't get panic messages when defmt-rtt; we need panic-probe to support RISCV/Xtensa.

Related: https://github.com/esp-rs/esp-backtrace/pull/40#issuecomment-1828154123

PR #76 (2023-12-15): Remove RTT, defmt-raw
Closes #65
closed 2023-12-15 by bjoernQ